STMicroelectronics/stm32ai-modelzoo-services #6: How can I increase the FPS of the model?

woshuhua, 2025-04-28:
I have tried to deploy the st_ssd_mobilenet_v1 for object detection on my STM32H747I-DISC0 board. The FPS for 025_192_int8.tflite is 6.6. How can I increase the FPS of the model for a real-time system? Can I adjust the clock rate to increase the FPS? If I wanna adjust the clock rate, I can't find any .ioc file inside the experiment_output directory, only network_config.h is generated inside the folder...

GRATTINSTM, 2025-05-22:
Hello, 
The STM32H747I-DISCO projects already use the maximum supported clock speed. The only way to get more FPS would be to turn to an another, more powerful MCU. Image processing on MCU is new and neural networks are particularly computationally intensive. [STM32N6](https://www.st.com/en/evaluation-tools/stm32n6570-dk.html) is specifically designed to run convolutional neural networks and is the best option to run computer vision models. 
Guillaume

woshuhua, 2025-06-08:
Thank you so much!

closed 2025-06-08 by woshuhua