STMicroelectronics/stm32ai-modelzoo-services #9: Error with deploy a model on the stm32n6

wyb217, 2025-05-13:
Hello,I met these logs when I try to deploy a model on stm32n6d570-dk:
```
1 physical GPUs, 1 logical GPUs
[INFO] : Setting upper memory limit to 4GBytes on gpu[0]
[INFO] : Running `deployment` operation mode
[INFO] : ClearML config check
[INFO] : The random seed for this simulation is 127
INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
[INFO] : Generating C header file for Getting Started...
loading model.. model_path="yolov8n_256_quant_pc_uf_od_coco-person-st.tflite"
loading conf file.. "../../application_code/object_detection/STM32N6/stmaic_STM32N6570-DK.conf" config="None"
"n6 release" configuration is used
[INFO] : Selected board :  "STM32N6570-DK Getting Started Object Detection (STM32CubeIDE)" (stm32_cube_ide/n6 release/stm32n6)
[INFO] : Compiling the model and generating optimized C code + Lib/Inc files:  yolov8n_256_quant_pc_uf_od_coco-person-st.tflite
setting STM.AI tools.. root_dir="", req_version=""
 Cube AI Path: "E:\ST\STEdgeAI\2.0\Utilities\windows\stedgeai.exe".
[INFO] : Offline CubeAI used; Selected tools:  10.0.0 (x-cube-ai pack)
loading conf file.. "../../application_code/object_detection/STM32N6/stmaic_STM32N6570-DK.conf" config="None"
"n6 release" configuration is used
compiling... "yolov8n_256_quant_pc_uf_od_coco-person-st_tflite" session
 model_path  : ['yolov8n_256_quant_pc_uf_od_coco-person-st.tflite']
 tools       : 10.0.0 (x-cube-ai pack)
 target      : "STM32N6570-DK Getting Started Object Detection (STM32CubeIDE)" (stm32_cube_ide/n6 release/stm32n6)
 options     : --st-neural-art default@../../application_code/object_detection/STM32N6/Model/user_neuralart.json --input-data-type uint8 --inputs-ch-position chlast
D:\PyCharm 2024.3.5\plugins\python-ce\helpers\pycharm_display\datalore\display\supported_data_type.py:6: UserWarning: The NumPy module was reloaded (imported a second time). This can in some cases result in small but subtle issues and is discouraged.
  import numpy
ST Edge AI Core v2.0.0-20049

Error executing job with overrides: []
Traceback (most recent call last):
  File "D:\code\stm32ai-modelzoo-services\object_detection\src\stm32ai_main.py", line 242, in <module>
    main()
  File "D:\anaconda\envs\st_zoo\lib\site-packages\hydra\main.py", line 94, in decorated_main
    _run_hydra(
  File "D:\anaconda\envs\st_zoo\lib\site-packages\hydra\_internal\utils.py", line 394, in _run_hydra
    _run_app(
  File "D:\anaconda\envs\st_zoo\lib\site-packages\hydra\_internal\utils.py", line 457, in _run_app
    run_and_report(
  File "D:\anaconda\envs\st_zoo\lib\site-packages\hydra\_internal\utils.py", line 223, in run_and_report
    raise ex
  File "D:\anaconda\envs\st_zoo\lib\site-packages\hydra\_internal\utils.py", line 220, in run_and_report
    return func()
  File "D:\anaconda\envs\st_zoo\lib\site-packages\hydra\_internal\utils.py", line 458, in <lambda>
    lambda: hydra.run(
  File "D:\anaconda\envs\st_zoo\lib\site-packages\clearml\binding\hydra_bind.py", line 91, in _patched_hydra_run
    return PatchHydra._original_hydra_run(self, config_name, task_function, overrides, *args, **kwargs)
  File "D:\anaconda\envs\st_zoo\lib\site-packages\hydra\_internal\hydra.py", line 132, in run
    _ = ret.return_value
  File "D:\anaconda\envs\st_zoo\lib\site-packages\hydra\core\utils.py", line 260, in return_value
    raise self._return_value
  File "D:\anaconda\envs\st_zoo\lib\site-packages\hydra\core\utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "D:\anaconda\envs\st_zoo\lib\site-packages\clearml\binding\hydra_bind.py", line 230, in _patched_task_function
    return task_function(a_config, *a_args, **a_kwargs)
  File "D:\code\stm32ai-modelzoo-services\object_detection\src\stm32ai_main.py", line 228, in main
    process_mode(cfg)
  File "D:\code\stm32ai-modelzoo-services\object_detection\src\stm32ai_main.py", line 102, in process_mode
    deploy(cfg)
  File "D:\code\stm32ai-modelzoo-services\object_detection\src\../deployment\deploy.py", line 111, in deploy
    stm32ai_deploy_stm32n6(target=board, stlink_serial_number=stlink_serial_number, stm32ai_version=stm32ai_version, c_project_path=c_project_path,
  File "D:\code\stm32ai-modelzoo-services\object_detection\src\../../common/deployment\common_deploy.py", line 469, in stm32ai_deploy_stm32n6
    stmaic_local_call(session)
  File "D:\code\stm32ai-modelzoo-services\object_detection\src\../../common/deployment\common_deploy.py", line 443, in stmaic_local_call
    stmaic.compile(session=session, options=opt, target=session._board_config)
  File "D:\code\stm32ai-modelzoo-services\object_detection\src\../../common\stm32ai_local\compile.py", line 208, in cmd_compile
    raise Exception('Error during compilation')
Exception: Error during compilation
```
I have tried to use the latest version of ai-core(2.1),but met the same error, why i met this error?I followed every step from[https://github.com/STMicroelectronics/stm32ai-modelzoo-services/blob/main/object_detection/deployment/README_STM32N6.md] .
Thanks very much!

fazenrazen, 2025-05-16:
Hello I would like to know where are your troubles, which step are you stuck specially? Seems like you are working with a local version of the stm model. Are you retraining a model?

wyb217, 2025-05-17:
@fazenrazen Hello, and thanks for offering to help!

I’m actually not sure exactly what’s going wrong—the build always fails with
Exception: Error during compilation.
I’ve pasted the full error output above.

My goal is simply to deploy a pretrained model onto the STM32 N6570-DK board using the automated .yaml–based workflow.
I use this model:[yolov8n_256_quant_pc_uf_od_coco-person-st.tflite](https://github.com/stm32-hotspot/ultralytics/blob/main/examples/YOLOv8-STEdgeAI/stedgeai_models/object_detection/yolov8n_256_quant_pc_uf_od_coco-person-st.tflite)  and the content of my .yaml file as follow:

 ```
general:
  project_name: COCO_2017_person_Demo
  model_type: yolo_v8
#choices=[st_ssd_mobilenet_v1, ssd_mobilenet_v2_fpnlite, tiny_yolo_v2, st_yolo_lc_v1, 
#         st_yolo_x, yolo_v8, yolo_v5u]
  model_path: yolov8n_256_quant_pc_uf_od_coco-person-st.tflite
  logs_dir: logs
  saved_models_dir: saved_models
  gpu_memory_limit: 6
  num_threads_tflite: 4
  global_seed: 127

operation_mode: deployment
#choices=['training' , 'evaluation', 'deployment', 'quantization', 'benchmarking',
#        'chain_tqeb','chain_tqe','chain_eqe','chain_qb','chain_eqeb','chain_qd ']

dataset:
  name: COCO_2017_person
  class_names: [ person ]
  training_path: /dataset/coco_person_2017_tfs/train
  validation_path:
  validation_split: 0.1
  test_path: /dataset/coco_person_2017_tfs/val
  quantization_path: /dataset/coco_person_2017_tfs/val
  quantization_split: 0.01

preprocessing:
  rescaling: { scale: 1/127.5, offset: -1 }
  resizing:
    aspect_ratio: fit
    interpolation: nearest
  color_mode: rgb
                       
data_augmentation:
  ########## For tiny_yolo_v2 and st_yolo_lc_v1 only ###########
  random_periodic_resizing:
    period: 10
    image_sizes: [(192, 192), (224, 224), (256, 256), (288, 288), (320, 320), (352, 352),
                  (384, 384), (416, 416), (448, 448), (480, 480), (512, 512),
                  (544, 544), (576, 576), (608, 608)]
  random_flip:
    mode: horizontal
  random_crop:
    crop_center_x: (0.25, 0.75)
    crop_center_y: (0.25, 0.75)
    crop_width: (0.5, 0.9)
    crop_height: (0.5, 0.9)
    change_rate: 0.9
  random_contrast:
    factor: 0.4
  random_brightness:
    factor: 0.3 

training:
  model:
    # alpha: 0.35
    input_shape: (192, 192, 3)
    # pretrained_weights: imagenet
  dropout:
  batch_size: 64
  epochs: 4
  optimizer:
    Adam:
      learning_rate: 0.005
  callbacks:
    ReduceLROnPlateau:
      monitor: val_map
      patience: 10
      factor: 0.25
    ModelCheckpoint:
      monitor: val_map
    EarlyStopping:
      monitor: val_map
      patience: 20

postprocessing:
  confidence_thresh: 0.1
  NMS_thresh: 0.5
  IoU_eval_thresh: 0.4
  plot_metrics: False   # Plot precision versus recall curves. Default is False.
  max_detection_boxes: 100

quantization:
  quantizer: TFlite_converter
  quantization_type: PTQ
  quantization_input_type: uint8
  quantization_output_type: float
  granularity: per_channel   #per_tensor
  optimize: False   #can be True if per_tensor
  export_dir: quantized_models

benchmarking:
  board: STM32H747I-DISCO

tools:
  stedgeai:
    version: 10.0.0
    optimization: balanced
    on_cloud: False
    path_to_stedgeai: E:/ST/STEdgeAI/2.0/Utilities/windows/stedgeai.exe
  path_to_cubeIDE: E:/ST/STM32CubeIDE_1.17.0/STM32CubeIDE/stm32cubeide.exe

deployment:
  c_project_path: D:\code\stm32ai-modelzoo-services\en.n6-ai-getstarted-v1.0.0\application_code\object_detection\STM32N6
  IDE: GCC
  verbosity: 1
  hardware_setup:
    serie: STM32N6
    board: STM32N6570-DK

mlflow:
  uri: ./experiments_outputs/mlruns

hydra:
  run:
    dir: ./experiments_outputs/${now:%Y_%m_%d_%H_%M_%S}
```
  Do you have some ideas about my trouble?Thanks very much!

fazenrazen, 2025-05-19:
I am having the same issue, I am looking into is, I do know it was working with ST EDGE AI 2.0, but do you know where the older files are? Cause in the 2.0 version you needed the NPU ART extension to add in when you install the application

fazenrazen, 2025-05-19:
I am still getting the same error with the 2.0 version of ST EDGE AI, I will try to reflash the boot using the ST programmer and then retry to flash the board. Kinda annoying if you ask me.

fazenrazen, 2025-05-19:
I cant seem to figure it out sadly. I am getting the same errors as you. In mine its saying that the outputed experiments are conflicting files but I have deleted it and its still causing issues. I think this repo doesnt work with ST EDGE 2.1 as of now. I would like @avisconti or @glegrain input on this. Thank you!

[image]

fazenrazen, 2025-05-20:
https://community.st.com/t5/edge-ai/i-m-currently-deploying-the-object-detection-of-stm32modelzoo/td-p/802776

@wyb217 they are fixing it this week! super exciting!

wyb217, 2025-05-20:
@fazenrazen Thank you for your enthusiastic help! My issue wasn’t the same as yours, but I also found the solution on the ST Forum:(https://community.st.com/t5/edge-ai/deploying-the-image-classification-model-on-the-stm32n6570-dk/m-p/792217) I sincerely appreciate your assistance!

fazenrazen, 2025-05-20:
[](https://discord.gg/cEYbQnd9) I have a discord channel if you would like to join!

wyb217, 2025-05-22:
@fazenrazen Thanks very much!I'd like to join!

fazenrazen, 2025-05-22:
send me ur discord or u can add my fazenrazen#6933 or just fazenrazen

fazenrazen, 2025-05-22:
sorry about that! i hope you have a good day :)

wyb217, 2025-05-22:
Thanks very much!I hope you have a great day too!

LFOSTM, 2025-05-22:
Hello,
The new version 3.1 of the model zoo should have fixed it.
Do you confirm this is OK?

wyb217, 2025-05-22:
@LFOSTM sorry I forget to close this issue, that's okay now.Thanks!

closed 2025-05-22 by wyb217